sandmmakers/BedLeveler5000 #21: No name 'StrEnum' in module 'enum' (no-name-in-module) / Cannot import 'SerialConnection' due to 'invalid syntax (<unknown>, line 32)' (syntax-error)

berlin4apk, 2023-11-28:
+ pylint --rcfile .pylintrc --errors-only src
************* Module Printers.Marlin2.SerialConnection
src/Printers/Marlin2/SerialConnection.py:32:15: E0001: Parsing failed: 'invalid syntax (<unknown>, line 32)' (syntax-error)
************* Module BedLeveler5000
src/BedLeveler5000.py:23:0: E0611: No name 'StrEnum' in module 'enum' (no-name-in-module)
************* Module Printers.CommandPrinter
src/Printers/CommandPrinter.py:10:18: E1101: Module 'enum' has no 'StrEnum' member (no-member)
************* Module Printers.Marlin2.LineConnection
src/Printers/Marlin2/LineConnection.py:3:0: E0001: Cannot import 'SerialConnection' due to 'invalid syntax (<unknown>, line 32)' (syntax-error)
************* Module Printers.Marlin2.CommandConnection
src/Printers/Marlin2/CommandConnection.py:18:0: E0001: Cannot import 'SerialConnection' due to 'invalid syntax (<unknown>, line 32)' (syntax-error)
************* Module Common.PrinterInfo
src/Common/PrinterInfo.py:6:0: E0611: No name 'StrEnum' in module 'enum' (no-name-in-module)

berlin4apk, 2023-11-28:
user@box:/tmp$ git clone https://github.com/sandmmakers/BedLeveler5000
Cloning into 'BedLeveler5000'...
remote: Enumerating objects: 374, done.
remote: Counting objects: 100% (374/374), done.
remote: Compressing objects: 100% (211/211), done.
remote: Total 374 (delta 229), reused 299 (delta 154), pack-reused 0
Receiving objects: 100% (374/374), 201.87 KiB | 2.69 MiB/s, done.
Resolving deltas: 100% (229/229), done.
user@box:/tmp$ cd BedLeveler5000
user@box:/tmp/BedLeveler5000$ python3 -m venv venv
user@box:/tmp/BedLeveler5000$ source venv/bin/activate
(venv) user@box:/tmp/BedLeveler5000$ pip3 install -r requirements.txt
Collecting PySide6
  Using cached PySide6-6.6.0-cp38-abi3-manylinux_2_28_x86_64.whl (6.7 kB)
Collecting pyinstaller
  Using cached pyinstaller-6.2.0-py3-none-manylinux2014_x86_64.whl (663 kB)
Collecting pillow
  Using cached Pillow-10.1.0-cp39-cp39-manylinux_2_28_x86_64.whl (3.6 MB)
Collecting pylint
  Using cached pylint-3.0.2-py3-none-any.whl (510 kB)
Collecting altgraph
  Using cached altgraph-0.17.4-py2.py3-none-any.whl (21 kB)
Collecting pyinstaller-hooks-contrib>=2021.4
  Using cached pyinstaller_hooks_contrib-2023.10-py2.py3-none-any.whl (288 kB)
Collecting packaging>=22.0
  Using cached packaging-23.2-py3-none-any.whl (53 kB)
Collecting importlib-metadata>=4.6
  Using cached importlib_metadata-6.8.0-py3-none-any.whl (22 kB)
Requirement already satisfied: setuptools>=42.0.0 in ./venv/lib/python3.9/site-packages (from pyinstaller->-r requirements.txt (line 2)) (44.1.1)
Collecting zipp>=0.5
  Using cached zipp-3.17.0-py3-none-any.whl (7.4 kB)
Collecting astroid<=3.1.0-dev0,>=3.0.1
  Using cached astroid-3.0.1-py3-none-any.whl (275 kB)
Collecting tomlkit>=0.10.1
  Using cached tomlkit-0.12.3-py3-none-any.whl (37 kB)
Collecting dill>=0.2
  Using cached dill-0.3.7-py3-none-any.whl (115 kB)
Collecting mccabe<0.8,>=0.6
  Using cached mccabe-0.7.0-py2.py3-none-any.whl (7.3 kB)
Collecting typing-extensions>=3.10.0
  Using cached typing_extensions-4.8.0-py3-none-any.whl (31 kB)
Collecting tomli>=1.1.0
  Using cached tomli-2.0.1-py3-none-any.whl (12 kB)
Collecting platformdirs>=2.2.0
  Using cached platformdirs-4.0.0-py3-none-any.whl (17 kB)
Collecting isort<6,>=4.2.5
  Using cached isort-5.12.0-py3-none-any.whl (91 kB)
Collecting PySide6-Essentials==6.6.0
  Downloading PySide6_Essentials-6.6.0-cp38-abi3-manylinux_2_28_x86_64.whl (82.1 MB)
     |████████████████████████████████| 82.1 MB 76 kB/s 
Collecting PySide6-Addons==6.6.0
  Downloading PySide6_Addons-6.6.0-cp38-abi3-manylinux_2_28_x86_64.whl (125.2 MB)
     |████████████████████████████████| 125.2 MB 518 kB/s 
Collecting shiboken6==6.6.0
  Downloading shiboken6-6.6.0-cp38-abi3-manylinux_2_28_x86_64.whl (169 kB)
     |████████████████████████████████| 169 kB 3.0 MB/s 
Installing collected packages: shiboken6, zipp, typing-extensions, PySide6-Essentials, tomlkit, tomli, PySide6-Addons, pyinstaller-hooks-contrib, platformdirs, packaging, mccabe, isort, importlib-metadata, dill, astroid, altgraph, PySide6, pylint, pyinstaller, pillow
Successfully installed PySide6-6.6.0 PySide6-Addons-6.6.0 PySide6-Essentials-6.6.0 altgraph-0.17.4 astroid-3.0.1 dill-0.3.7 importlib-metadata-6.8.0 isort-5.12.0 mccabe-0.7.0 packaging-23.2 pillow-10.1.0 platformdirs-4.0.0 pyinstaller-6.2.0 pyinstaller-hooks-contrib-2023.10 pylint-3.0.2 shiboken6-6.6.0 tomli-2.0.1 tomlkit-0.12.3 typing-extensions-4.8.0 zipp-3.17.0
(venv) user@box:/tmp/BedLeveler5000$ ./build_all
bash: ./build_all: Permission denied
(venv) user@box:/tmp/BedLeveler5000$ bash ./build_all
+++ dirname ./build_all
++ cd .
++ pwd
+ TOP_LEVEL_DIR=/tmp/BedLeveler5000
+ BUILD_DIR=/tmp/BedLeveler5000/Build
+ INSTALL_DIR=/tmp/BedLeveler5000/Install
+ rm -rf /tmp/BedLeveler5000/Build
+ rm -rf /tmp/BedLeveler5000/Install
+ pylint --rcfile .pylintrc --errors-only src
************* Module Printers.Marlin2.SerialConnection
src/Printers/Marlin2/SerialConnection.py:32:15: E0001: Parsing failed: 'invalid syntax (<unknown>, line 32)' (syntax-error)
************* Module BedLeveler5000
src/BedLeveler5000.py:23:0: E0611: No name 'StrEnum' in module 'enum' (no-name-in-module)
************* Module Printers.CommandPrinter
src/Printers/CommandPrinter.py:10:18: E1101: Module 'enum' has no 'StrEnum' member (no-member)
************* Module Printers.Marlin2.LineConnection
src/Printers/Marlin2/LineConnection.py:3:0: E0001: Cannot import 'SerialConnection' due to 'invalid syntax (<unknown>, line 32)' (syntax-error)
************* Module Printers.Marlin2.CommandConnection
src/Printers/Marlin2/CommandConnection.py:18:0: E0001: Cannot import 'SerialConnection' due to 'invalid syntax (<unknown>, line 32)' (syntax-error)
************* Module Common.PrinterInfo
src/Common/PrinterInfo.py:6:0: E0611: No name 'StrEnum' in module 'enum' (no-name-in-module)
(venv) user@box:/tmp/BedLeveler5000$

sandmmakers, 2023-11-28:
Can you please check your version of Python `python3 --version`? I think at least Python 3.10 is required.

berlin4apk, 2023-11-28:
user@box:~$ python3 --version
Python 3.9.2
user@box:~$

closed 2023-11-28 by sandmmakers