sandmmakers/BedLeveler5000 #29: Error probe

maj0r1337, 2023-11-30:
hi,
thank you for that nice tool!
ive tried to make my own config for the neptune 4 pro. my problem is, if i connect the printer the homing process works great but if i klick on any button 1-4 or h i recive an error "probe"

[image]

here is my json for the 4 pro

> {
    "version": 2,
    "displayName": "Neptune 4 Pro",
    "connectionMode": "Moonraker",
    "connection": {},
    "manualProbePoints": [{
                              "name": "vorne links",
                              "row": 2,
                              "column": 0,
                              "x": 47.0,
                              "y": 12.0
                          },
                          {
                              "name": "hinten links",
                              "row": 0,
                              "column": 0,
                              "x": 47.0,
                              "y": 187.0
                          },
                          {
                              "name": "vorne rechts",
                              "row": 2,
                              "column": 2,
                              "x": 219.0,
                              "y": 12.0
                          },
                          {
                              "name": "hinten rechts",
                              "row": 0,
                              "column": 2,
                              "x": 219.0,
                              "y": 187.0
                          },
                          {
                              "name": "home",
                              "row": 1,
                              "column": 1,
                              "x": 141.75,
                              "y": 97.05
                          }
                         ]
}

sandmmakers, 2023-11-30:
Can you please attach your `printer.cfg` file?

sandmmakers, 2023-12-01:
Thank you. I'll put this in my simulator tonight and try to reproduce the error.

maj0r1337, 2023-12-01:
Thank you, hope youl find the error, i like the tool a lot =)

maj0r1337, 2023-12-03:
yes, a mesh is loaded
[image]

sandmmakers, 2023-12-04:
A new version of the Bed Leveler 5000 Software Suite has been released which should resolve this issue. If it doesn't, or there are any new issues, the documentation has been updated with steps to create a log file.

maj0r1337, 2023-12-04:
Thank you for the Update
I have now the following error code

[image]

sandmmakers, 2023-12-04:
Did you use the **Printer Info Wizard** to create your configuration file? While I didn't look at the numbers in your configuration file before, they might be incorrect since they match your `screws_tilt_adjust` numbers. However, the two sets of numbers should not match. The BL5K configuration file contains the coordinates to probe while the `screws_tilt_adjust` numbers are the locations of the nozzle during probing. If the `screws_tilt_adjust` numbers are used in the BL5K file, then the XY probe offset would be  applied twice.

If your `screws_tilt_adjust` numbers are correct, then the coordinates in the BL5K configuration file should be:
22.75, 32.45
22.75, 207.45
194.75, 207.45
194.75, 32.45

However, since the numbers do not end in `.0` or `.5`, there might be deviations from the values used by the LCD controller when you press the manual level buttons. If you want BL5K to probe the exact locations where the LCD places the nozzle during manual bed leveling, you might want to use the **query** feature in the **Printer Info Wizard**.

sandmmakers, 2023-12-04:
BL5K gets all printer information directly from the printer, nothing is hard coded into the source code.

You values in your new configuration files don't seem to be correct. When you used PIW, how did you position the toolhead? PIW queries the location of the nozzle, not the probe, so the nozzle needs to be over the intended location. This is because when you use the LCD controller, the nozzle is placed over the location to be tested with paper. Sorry for the confusion. 

Tonight, I will change PIW so there are two query buttons, one for querying the nozzle location and one for the probe location. Also, since I haven't released the Klipper support in PIW video yet, I will redo the raw footage to make this all very clear.

Thank you for all of your help working through all of these BL5K growing pains!!!!!