sandmmakers/BedLeveler5000 #37: Won't install

TwistedPrints3D, 2023-12-14:
I have tried to install this on windows 11, I keep getting a error message. i have added a screenshot  of the error code

[image]

sandmmakers, 2023-12-15:
There are no installers in the BL5K software suite. When you launch one of the applications, the bootloader extracts some files to a temporary directory and sets up a Python environment, then starts the Python code. Once the application is closed, the environment and temp files are removed.

The described error happens before control reaches any BL5K code, so it must be related the BL5K software suite using PyInstaller to bundle the Python code in executables. Do you have an anti-virus program running? Can you check if the file referenced in the error message was quarantined? There are no viruses in the BL5K software suite, but it can get flagged as a false positive because of its use of PyInstaller ( I'm currently working on a workaround, but it's not ready yet).

TwistedPrints3D, 2023-12-15:
I turned off all anti virus software and same thing. Should I try a different version?

sandmmakers, 2023-12-15:
Many people have been able to use BL5K without any issues so either there is something different about your system or the archive/files are corrupt. Can you please try redownloading and trying again? Thank you.

TwistedPrints3D, 2023-12-15:
I understand that. I’ve seen many on fb using it. I’ve deleted all the downloads and redownloaded it and still the same thing. My computer is only 3 weeks old so it’s possible that something in my system is stopping it. I’m going to try my sons computer

TwistedPrints3D, 2023-12-17:
I got it working. It was 100% user error. Had someone from the elegoo neptune 4 page on facebook help me figure out what I did wrong. I forgot to move it out of the 7-zip file onto my desktop and open it there. I was trying to open it from 7-zip. Thank you for all your help.

sandmmakers, 2023-12-17:
I'm glad you got it working!

closed 2023-12-17 by sandmmakers