sandmmakers/BedLeveler5000 #40: Contains

Thebytebeyond, 2023-12-30:


sandmmakers, 2023-12-30:
Thank you for your interest in BL5K. These are false positives caused by overly aggressive and error-prone AI hueristics. In the past, some "script kiddies" used PyInstaller to bundle malware, and now, as a result, a number of anti-virus programs flag anything created with PyInstaller. Please updated your virus definitions.

[image]

sandmmakers, 2023-12-30:
Microsoft has white-listed release 0.4.0 for a second time: 
[image]

sandmmakers, 2023-12-30:
Microsoft has white listed release v0.4.0 for a second time. Hopefully, this time it actually works.
[image]

sandmmakers, 2023-12-31:
With the latest set of Windows Defender virus definitions, no false positives are currently being observed.

closed 2023-12-31 by sandmmakers

Thebytebeyond, 2024-01-03:
Win defender doesnt flag and malwarebytes doesnt flag. thanks for the hard work @sandmmakers

sandmmakers, 2024-01-03:
Thank you for using BL5K!!