sandmmakers/BedLeveler5000 #41: Question for Neptune 4

ygarand, 2024-01-03:
Hi,

I have an Elegoo Neptune 4 can we used the config for Neptune 4 Pro for the leveling.

Thanks

sandmmakers, 2024-01-03:
While using a configuration file for a different model printer (but same firmware type) will technically work, the manual probe locations will be wrong. You can use the Printer Info Wizard to easily and quickly create a configuration file for a new printer. However, a user has recently provided a configuration file for the Neptune 4 and I can commit it either tonight or tomorrow. While it will be included in the next release, I'm holding off creating a new release until there are a couple of new features added since dealing with anti-virus false positives is such a pain.

ygarand, 2024-01-03:
Thanks for the answer.

Okay I will wait the next update.

Thanks

Yannick Garand Envoyé de mon iPhone

sandmmakers, 2024-01-03:
While a new release has not been created yet, a Neptune 4 configuration file was just merged in and you can download it directly (or use git if you cloned the repo). This is a user contributed configuration file and has not been tested.

closed 2024-01-03 by sandmmakers

ygarand, 2024-02-06:
Hi,

I downloaded the last version of the Bed Leveler 5000 but wi don't find the config for Neptune 4. Can you added when you have a moment. 

Thanks

sandmmakers, 2024-02-06:
The configuration file will be included in the next release, but for right now, you can download it from:
https://github.com/sandmmakers/BedLeveler5000/tree/master/Printers