sandmmakers/BedLeveler5000 #48: Neptune 4 Pro - Bad Request

ChuckMac, 2024-01-24:
Trying to run BL5K on my Neptune 4 Pro.  I can connect to the printer, set temperature, and probe point 1.  However when I attempt to probe any other points I get a "Bad Request" error and it disconnects from the printer.  Fluidd console is showing "Move out of range".

This is v0.4.0 on Windows

[image]

[image]

sandmmakers, 2024-01-24:
Thank you for creating the bug report. Can you please upload a log file? Directions are available here: https://github.com/sandmmakers/BedLeveler5000/blob/master/Docs/FilingABugReport.md

ChuckMac, 2024-01-24:
Thank you for the quick response. 

So after restarting in debug mode the issue went away.  Going to chalk it up to user error.

Thanks a lot for all your work on this!

closed 2024-01-24 by ChuckMac