sandmmakers/BedLeveler5000 #51: Manual Probe Neptune 3 Max (Klipper) incorrect location

Ctrl-Alt-Rage, 2024-02-07:
When I try to level the bed and click "1". Instead of moving to the front left corner, it moves about 6 inches to the right from the center and probes it. 

Then if I try to press "2" I get a movement out of range error. 

It's likely I'm doing it incorrectly, but am I missing something?

sandmmakers, 2024-02-07:
Please add your printer.cfg file. It sounds like your configuration might not be correct.

Ctrl-Alt-Rage, 2024-02-07:
Thanks for getting back to me so quickly. This is the printer.cfg that I have.

`[include mainsail.cfg]
[include area_bed_mesh.cfg]
[include KAMP_Settings.cfg]

# For the ZNP Robin Nano DW v2.2 board:
#   - Compile with the processor model STM32F401.
#   - Select the 32KiB bootloader,
#   - Select (Serial (on # USART1 PA10/PA9) for the communication interface.
#   - Select (Serial (on # USART2 PA3/PA2) if you want to use the serial UART pins on J17 (wifi section)
# Note that the "make flash" command does not work with ZNP Robin boards. After running "make", run the following command:
# cp out/klipper.bin out/ZNP_ROBIN_NANO.bin
#
# Copy the file out/ZNP_ROBIN_NANO.bin to an SD card and then restart the printer with that SD card.
# See docs/Config_Reference.md for a description of parameters.

#[include timelapse.cfg]

[exclude_object] #comment out for Sonic pad

[virtual_sdcard]
path: ~/printer_data/gcodes

[pause_resume]

[display_status]

[gcode_macro CANCEL_PRINT]
description: Cancel the actual running print
rename_existing: CANCEL_PRINT_BASE
gcode:
    TURN_OFF_HEATERS
    CANCEL_PRINT_BASE

[gcode_macro PAUSE]
description: Pause the actual running print
rename_existing: PAUSE_BASE
# change this if you need more or less extrusion
variable_extrude: 1.0
gcode:
    ##### read E from pause macro #####
    {% set E = printer["gcode_macro PAUSE"].extrude|float %}
    ##### set park positon for x and y #####
    # default is your max posion from your printer.cfg
    {% set x_park = printer.toolhead.axis_maximum.x|float - 5.0 %}
    {% set y_park = printer.toolhead.axis_maximum.y|float - 5.0 %}
    ##### calculate save lift position #####
    {% set max_z = printer.toolhead.axis_maximum.z|float %}
    {% set act_z = printer.toolhead.position.z|float %}
    {% if act_z < (max_z - 2.0) %}
        {% set z_safe = 2.0 %}
    {% else %}
        {% set z_safe = max_z - act_z %}
    {% endif %}
    ##### end of definitions #####
    PAUSE_BASE
    G91
    {% if printer.extruder.can_extrude|lower == 'true' %}
      G1 E-{E} F2100
    {% else %}
      {action_respond_info("Extruder not hot enough")}
    {% endif %}
    {% if "xyz" in printer.toolhead.homed_axes %}
      G1 Z{z_safe} F900
      G90
      G1 X{x_park} Y{y_park} F6000
    {% else %}
      {action_respond_info("Printer not homed")}
    {% endif %} 
[... 231 lines cut ...]
#*# pid_kd = 118.145
#*#
#*# [heater_bed]
#*# control = pid
#*# pid_kp = 72.306
#*# pid_ki = 0.844
#*# pid_kd = 1548.255
#*#
#*# [bed_mesh default]
#*# version = 1
#*# points =
#*# 	  0.025833, 0.005833, 0.015000
#*# 	  0.025000, 0.006667, 0.014167
#*# 	  0.002500, -0.026667, -0.016667
#*# x_count = 3
#*# y_count = 3
#*# mesh_x_pps = 2
#*# mesh_y_pps = 2
#*# algo = lagrange
#*# tension = 0.2
#*# min_x = 177.54
#*# max_x = 230.91000000000003
#*# min_y = 182.83
#*# max_y = 241.62
`

Ctrl-Alt-Rage, 2024-02-07:
1 - The LCD Controller has been unplugged since I installed Klipper due to some reported start up issues. That could have been fixed since then (I got the printer beginning of December). The only thing currently connected to the printer is the PI box to connect to it. 

2 - I haven't tried restarting the printer or turning it off and on again. 

I will report any findings later tonight or tomorrow. It's in the middle of a print job right now. 

Thanks again!

sandmmakers, 2024-02-07:
Thank you for the update. In the mean time, I will test in the simulator.

sandmmakers, 2024-02-07:
I just confirmed that BL5K does not ensure the printer is in absolute positioning mode. While there is a chance there is something else causing the behavior you observed, my guess is that you ran something, or did something, that left the printer in relative positioning mode. I will fix this for the next release.

sandmmakers, 2024-02-07:
I just pushed a fix. You can see the changes here: https://github.com/sandmmakers/BedLeveler5000/commit/148ea309f2807dce1ed8d4b67e00106f541b315c.

While it might be best to just restart your printer and test again with v0.4.0, you can test a preview build here: https://github.com/sandmmakers/BedLeveler5000/actions/runs/7809227483.  This build has a lot of unrelated changes in it and is not fully tested yet.

Ctrl-Alt-Rage, 2024-02-07:
I just wanted to let you know (prior to installing your update above) doing a firmware restart in Klipper fixed the issue.

Thank you!

closed 2024-02-07 by sandmmakers