sandmmakers/BedLeveler5000 #53: MacOS Unable to Connect to Printer

RACharlesworth, 2024-02-19:
When attempting to connect to my printer, the "Initializing printer" dialog displays and nothing else happens.  In the terminal, the following is displayed:

Traceback (most recent call last):
  File "Printers/Moonraker/MoonrakerPrinter.py", line 188, in processReply
  File "json/__init__.py", line 346, in loads
  File "json/decoder.py", line 337, in decode
  File "json/decoder.py", line 355, in raw_decode
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)
Traceback (most recent call last):
  File "Printers/Moonraker/MoonrakerPrinter.py", line 188, in processReply
  File "json/__init__.py", line 346, in loads
  File "json/decoder.py", line 337, in decode
  File "json/decoder.py", line 355, in raw_decode
json.decoder.JSONDecodeError: Expecting value: line 1 column 1 (char 0)

I have already re-downloaded the file and tried again.
I have also tried to remove all printers from the "Printers" folder, other than the N4+ and still getting the same result.

Computer:

2023 Apple MacBook Pro
Apple M2 Pro chip
MacOS Sonoma v. 14.1.2

Printer:

Elegoo Neptune 4 Plus

[image]

[image]

sandmmakers, 2024-02-19:
I misread the error message when looking at it on my phone. This looks like there is something wrong or different with the responses coming from the printer. Since the particular line referenced parses every response from the printer, I'm not able to tell exactly which message is causing the error. However, I suspect there is something unexpected or different in the printers `printer.cfg` file. Can you please attach your `printer.cfg` file and also attach a log file? Those will let me see exactly what is causing the parser to fail. My best guess right now is one or more unicode characters are in your `printer.cfg` file, but that is just a guess.

sandmmakers, 2024-02-19:
I forgot to mention it in my previous response. The directions for creating a log file are in the "How to file a bug" documentation page.

sandmmakers, 2024-02-21:
Hello. Any updates? Thanks

RACharlesworth, 2024-02-26:
Sorry, it took me some time to try to figure out where to find the config file, I assume this is it I have copied it from Fluidd:

####################################################################################
# Machine type: NEPTUNE 4 PLUS
# Current configuration version: V1.4
# Date：2023-11-17
####################################################################################
# 本产品（Neptune 4/4pro/4plus/4max）使用自定义的klipper固件，不支持用户自行
# 更新官方Klipper/Fluidd/Moonraker，否则会导致机器无法正常使用。如果
# 您是这方面的专家或者对这方面感兴趣，希望自行DIY，我们建议先自备烧
# 录镜像的工具，以防止意外导致机器无法正常使用时可以进行恢复，烧录镜像
# 教程请联系售后支持团队获取，感谢您的配合！
####################################################################################
# This product (Neptune 4/4pro/4plus/4max) adopts Klipper firmware 
# and does not support users to update the official klipper/fluidd
# /moonraker by themselves, otherwise the machine will not work properly.
# If you want to DIY and are an expert or interested in this field, 
# we recommend that you prepare your own tools for burning images so 
# that you can restore them if problems arise. Please contact after-sales 
# support team for tutorials on burning images. Thank you for your cooperation!
####################################################################################
# This file contains common pin mappings for ZNP-K1-V2.0 
# boards. To use this config, the firmware should be compiled for the
# STM32F402. When running "make menuconfig", enable "extra low-level
# configuration setup", select the 32KiB bootloader, and serial (on
# USART1 PA10/PA9) communication.

# The "make flash" command does not work on the ZNP-K1-V2.0. Instead,
# after running "make", copy the generated "out/klipper.bin" file to a
# file named "elegoo_k1.bin" on an SD card and then restart the ZNP-K1-V2.0
# with that SD card.

# See docs/Config_Reference.md for a description of parameters.
####################################################################################

[include plr.cfg]

[mcu]
# The hardware use USART1 PA10/PA9 connect to RK3328
serial: /dev/ttyS0
restart_method: command

[stepper_x]
step_pin:PC14
dir_pin:PC13
enable_pin:!PC15
microsteps: 16
rotation_distance: 40
full_steps_per_rotation:200
endstop_pin:tmc2209_stepper_x:virtual_endstop
position_min: -12  
position_endstop: -10
position_max: 325
homing_speed:50
homing_retract_dist:0
homing_positive_dir:false

[stepper_y]
step_pin:PB4
dir_pin:!PB3
enable_pin:!PC15
microsteps:16
rotation_distance: 40
full_steps_per_rotation:200
endstop_pin:tmc2209_stepper_y:virtual_endstop
position_min: -2
position_endstop:0 
position_max:325
homing_speed:50
homing_retract_dist:0
[... 682 lines cut ...]

[include moonraker_obico_macros.cfg]

#*# <---------------------- SAVE_CONFIG ---------------------->
#*# DO NOT EDIT THIS BLOCK OR BELOW. The contents are auto-generated.
#*#
#*# [bed_mesh 6]
#*# version = 1
#*# points =
#*# 	  0.283750, 0.148750, 0.058750, 0.046250, 0.110000, 0.201250
#*# 	  0.255000, 0.106250, 0.006250, -0.012500, 0.067500, 0.115000
#*# 	  0.226250, 0.110000, 0.027500, -0.003750, 0.072500, 0.105000
#*# 	  0.180000, 0.080000, 0.012500, -0.011250, 0.060000, 0.081250
#*# 	  0.178750, 0.071250, -0.003750, -0.028750, 0.056250, 0.103750
#*# 	  0.193750, 0.078750, -0.011250, -0.030000, 0.047500, 0.113750
#*# tension = 0.2
#*# min_x = 10.0
#*# algo = bicubic
#*# y_count = 6
#*# mesh_y_pps = 2
#*# min_y = 21.0
#*# x_count = 6
#*# max_y = 315.0
#*# mesh_x_pps = 2
#*# max_x = 300.0

RACharlesworth, 2024-02-26:
And the BL5000 log file:
2024-02-19 13:56:22,130 - Bed Leveler 5000 - INFO - Starting Bed Leveler 5000
2024-02-19 13:56:27,800 - Printer - INFO - open(host=)
2024-02-19 13:56:27,801 - CommandPrinter - INFO - init(init-0)
2024-02-19 13:56:27,802 - root - DEBUG - Starting <class 'Printers.Moonraker.MoonrakerPrinter.InitMachine'> with id: init-0, context: None
2024-02-19 13:56:27,802 - InitMachine - DEBUG - Sending get: http:///printer/gcode/script?script=G28
2024-02-19 13:56:28,853 - CommandPrinter - INFO - getTemperatures(getTemperatures-1)
2024-02-19 13:56:28,854 - root - DEBUG - Starting <class 'Printers.Moonraker.MoonrakerPrinter.GetTemperaturesMachine'> with id: getTemperatures-1, context: None
2024-02-19 13:56:28,854 - GetTemperaturesMachine - DEBUG - Sending get: http:///printer/objects/query?heater_bed&extruder
2024-02-19 13:56:33,957 - Printer - INFO - close()

sandmmakers, 2024-03-08:
I'm assuming filling in the host field resolved your issue. If you are still experiencing problems, please open another ticket.

closed 2024-03-08 by sandmmakers