sandmmakers/BedLeveler5000 #54: Error Message : Incorrect number of tokens in response

FR009231, 2024-02-19:
[BedLeveler5000.exe --log-level debug --log-file log.txt](https://github.com/sandmmakers/BedLeveler5000/files/14331492/BedLeveler5000.exe.--log-level.debug.--log-file.log.txt)
[Artillery X1.json](https://github.com/sandmmakers/BedLeveler5000/files/14331504/Artillery.X1.json)

I'm using a Artillery Sidewinder X1 Printer.
I made my own configuration file to use it in BL5K.
after clicking on connect , printer start, and got the error message :  Incorrect number of tokens in response.

sandmmakers, 2024-02-19:
Thank you for creating the issue. The first file you attached doesn't appear to be a log file. Can you please check. I think it might be the Bed Leveler 5000 executable.

Also, you might want to set your center manual probe point to fixed (fixed means "not adjustable with a screw").

FR009231, 2024-02-19:
[log.txt](https://github.com/sandmmakers/BedLeveler5000/files/14331853/log.txt)
sorry :-)

FR009231, 2024-02-19:
[Configuration.txt](https://github.com/sandmmakers/BedLeveler5000/files/14331945/Configuration.txt)
this is my printer configuration

sandmmakers, 2024-02-19:
I think I see the problem. The version of Marlin running on your printer appears to have a completely different response format for the M851 command. Do you happen to know the version of Marlin? I doubt I can get older versions of Marlin running in the simulator, but I can look through the source in case there are any other differences. Fixing this issue will require a code change.

FR009231, 2024-02-19:
Marlin V1.1.9,
Arduino 1.8.19

FR009231, 2024-02-19:
https://github.com/artillery3d/sidewinder-x1-firmware

sandmmakers, 2024-02-19:
Your printer is using Marlin 1, which I haven't worked with before. Looking at the code and your log, adding compatibility might not be a quick fix because the commands for Marlin 1 appear to have a lot of differences from Marlin 2. I can get it to work, but unless I can get Marlin 1.x running on one of my printers, I might need a lot of log files.

FR009231, 2024-02-20:
I can understand how challenging it could be. If you need I can provide you all the log files you will need :-). I think your BL5K software could be the best one! I'm using Pronterface which is not so user friendly compared to your solution. just tell me if you would like to get my support. Greetings from Germany

sandmmakers, 2024-03-04:
After giving it a lot of thought and consideration, I don't think I will be able to add Marlin 1 support at this time. While technically possible, it would require a significant amount of effort and only a very small number of printers would be able to use the new support. Printers would need to have at least Marlin 1.1.2, a Z-probe, have not been updated to Marlin 2, and still be working.

closed 2024-03-04 by sandmmakers