sandmmakers/BedLeveler5000 #55: Error M420

altcontroldelete22, 2024-02-19:
Trying to connect to my Neptune 3 Max with BL5 and i am getting a error message failed to parse the m420 output
[image]
[image]

sandmmakers, 2024-02-19:
Thank you for creating the ticket. Can you please create a log file and add it to this ticket? The directions for creating a log file are in the "How to file a bug?" page int eh documentation. Without a log file, I can't really say what is causing the problem.

sandmmakers, 2024-02-19:
I just noticed that you are using an older version of BL5K. Please try using version v0.5.0. I think the issue might be that you don't have a mesh loaded in your printer. BL5K used to have a descriptive error message for this situation, but there was a regression in version v0.4.0, where the error was properly detected, but the error message wasn't as helpful as it could be. Version v0.5.0 restores the more helpful error message.

altcontroldelete22, 2024-02-20:
Hello, thank you for your timely response. Attached is the log file: hxxps[:]//pastebin[.]com/rQygvaGC

altcontroldelete22, 2024-02-20:
[log.log](https://github.com/sandmmakers/BedLeveler5000/files/14338105/log.log)

altcontroldelete22, 2024-02-20:
[image]

sandmmakers, 2024-02-20:
Thank you for the log and trying version v0.5.0. It appears that your printer currently doesn't have a mesh. Please run ABL and then try again.

altcontroldelete22, 2024-02-20:
im getting the same error message
just ran abl twice

sandmmakers, 2024-02-20:
You performed ABL using the printer's LCD controller?

altcontroldelete22, 2024-02-20:
yes sir

sandmmakers, 2024-02-20:
Did you disconnect BL5K before performing ABL and then reconnect after ABL was complete? If so, then something is very strange.

altcontroldelete22, 2024-02-20:
give me a second , going to re run it now

sandmmakers, 2024-02-20:
In the provided log file, your printer is reporting that the mesh is either missing or bad. You can test it by connecting to your printer with the Inspector G-code utility and issuing a `M420 V` command.

altcontroldelete22, 2024-02-20:
[image]

sandmmakers, 2024-02-20:
I'm creating a mesh on my Neptune 3 Max and then I will test just to make sure the code is working properly. However, my guess is that there is something wrong with your printer.

altcontroldelete22, 2024-02-20:
Ok thank you.

sandmmakers, 2024-02-20:
I powered on my printer, went to the leveling screen, clicked measure, let ABL complete, navigated back to the main/home screen, and then issues the `M420 V` command in Inspector G-code and got the expected mesh output.

sandmmakers, 2024-02-20:
This is what the output of the `M420 V ` command should be: 
[image]

sandmmakers, 2024-02-20:
When you perform ABL, do all of the mesh values appear on the LCD screen? If you are not able to create a mesh then I'm really not sure what else to suggest except to contact Elegoo at 3dp@elegoo.com. A this point, we are just testing the Elegoo provided firmware and since it is Marlin, there is really nothing to tweak or change.

altcontroldelete22, 2024-02-20:
no it stops at the 4th probe and gives me the probing failed screen on the lcd

altcontroldelete22, 2024-02-20:
yes i just contacted them,,, unbelievable

altcontroldelete22, 2024-02-20:
[image]

altcontroldelete22, 2024-02-21:
allright !!

altcontroldelete22, 2024-02-21:
[image]

altcontroldelete22, 2024-02-21:
[image]

sandmmakers, 2024-02-21:
Fantastic! You got it to work!! What was the problem?

closed 2024-02-22 by sandmmakers