sandmmakers/BedLeveler5000 #56: Add a step between connect and initializing printer

spikeon, 2024-02-21:
When connecting to the printer the first thing your software does is send commands to it that move the print bed.  It is not clearly communicated before clicking connect that this will absolutely destroy any currently running print job. 

I was trying to download and setup this software in preparation of running it after my current print.  My plan was to get it all setup and ready to go, so I connected it to my printer. Carnage ensued.

sandmmakers, 2024-02-21:
BL5K is not intended to be used while any other software is connected to the printer. Viewing the printer status in Fluidd or Mainsail is fine, though. Given the design of BL5K and how Marlin and Klipper work, there really isn't a separate "connect" stage, just the required initialization steps. I'm not sure if it is even possible to detect if another application is sending commands to the printer, but I can look into it. Maybe, if there is a reliable way to detect if the printer is already "in use", BL5K can display an error message.

spikeon, 2024-02-21:
I would suggest either adding a step between "connect" and "do anything to the printer" or have it display a warning message before connect that informs the user of what you just said.