sandmmakers/BedLeveler5000 #60: Fixed reference reference keeps increasing without touching the levelknobs.

Toetje585, 2024-03-15:
I have a Neptune 3 Plus (Klipper), and for some reason the Fixed reference point keeps increasing the test done below are without touching the level springs at all. What could be the cause of this?

Thanks for the great tool!

```
Probed 1 ( 37.5,  32.5):  1.273 (Adjust: 00:11 CW)
Probed 2 ( 37.5, 165.0):  1.271 (Adjust: 00:12 CW)
Probed 3 ( 37.5, 292.5):  1.259 (Adjust: 00:13 CW)
Probed 4 (292.5, 292.5):  1.290 (Adjust: 00:09 CW)
Probed 5 (292.5, 165.0):  1.298 (Adjust: 00:08 CW)
Probed 6 (292.5,  32.5):  1.271 (Adjust: 00:12 CW)
Probed H (160.8, 166.6):  1.368 (Fixed reference)
--------------------------------------------------
Probed 1 ( 37.5,  32.5):  1.327 (Adjust: 00:11 CW)
Probed 2 ( 37.5, 165.0):  1.309 (Adjust: 00:13 CW)
Probed 3 ( 37.5, 292.5):  1.305 (Adjust: 00:13 CW)
Probed 4 (292.5, 292.5):  1.331 (Adjust: 00:10 CW)
Probed 5 (292.5, 165.0):  1.340 (Adjust: 00:09 CW)
Probed 6 (292.5,  32.5):  1.324 (Adjust: 00:11 CW)
Probed H (160.8, 166.6):  1.414 (Fixed reference)
--------------------------------------------------
Probed 1 ( 37.5,  32.5):  1.375 (Adjust: 00:08 CW)
Probed 2 ( 37.5, 165.0):  1.358 (Adjust: 00:10 CW)
Probed 3 ( 37.5, 292.5):  1.352 (Adjust: 00:11 CW)
Probed 4 (292.5, 292.5):  1.382 (Adjust: 00:07 CW)
Probed 5 (292.5, 165.0):  1.388 (Adjust: 00:06 CW)
Probed 6 (292.5,  32.5):  1.362 (Adjust: 00:10 CW)
Probed H (160.8, 166.6):  1.441 (Fixed reference)
--------------------------------------------------
Probed 1 ( 37.5,  32.5):  1.421 (Adjust: 00:07 CW)
Probed 2 ( 37.5, 165.0):  1.397 (Adjust: 00:10 CW)
Probed 3 ( 37.5, 292.5):  1.397 (Adjust: 00:10 CW)
Probed 4 (292.5, 292.5):  1.428 (Adjust: 00:06 CW)
Probed 5 (292.5, 165.0):  1.415 (Adjust: 00:07 CW)
Probed 6 (292.5,  32.5):  1.407 (Adjust: 00:08 CW)
Probed H (160.8, 166.6):  1.477 (Fixed reference)
--------------------------------------------------
Probed 1 ( 37.5,  32.5):  1.448 (Adjust: 00:06 CW)
Probed 2 ( 37.5, 165.0):  1.423 (Adjust: 00:09 CW)
Probed 3 ( 37.5, 292.5):  1.432 (Adjust: 00:08 CW)
Probed 4 (292.5, 292.5):  1.450 (Adjust: 00:06 CW)
Probed 5 (292.5, 165.0):  1.442 (Adjust: 00:07 CW)
Probed 6 (292.5,  32.5):  1.428 (Adjust: 00:09 CW)
Probed H (160.8, 166.6):  1.500 (Fixed reference)
--------------------------------------------------
Probed 1 ( 37.5,  32.5):  1.466 (Adjust: 00:06 CW)
Probed 2 ( 37.5, 165.0):  1.444 (Adjust: 00:09 CW)
Probed 3 ( 37.5, 292.5):  1.453 (Adjust: 00:08 CW)
Probed 4 (292.5, 292.5):  1.473 (Adjust: 00:05 CW)
Probed 5 (292.5, 165.0):  1.457 (Adjust: 00:07 CW)
Probed 6 (292.5,  32.5):  1.445 (Adjust: 00:09 CW)
Probed H (160.8, 166.6):  1.516 (Fixed reference)
--------------------------------------------------
Probed 1 ( 37.5,  32.5):  1.526 (Adjust: 00:05 CW)
Probed 2 ( 37.5, 165.0):  1.503 (Adjust: 00:08 CW)
Probed 3 ( 37.5, 292.5):  1.522 (Adjust: 00:06 CW)
Probed 4 (292.5, 292.5):  1.538 (Adjust: 00:04 CW)
Probed 5 (292.5, 165.0):  1.509 (Adjust: 00:07 CW)
Probed 6 (292.5,  32.5):  1.513 (Adjust: 00:07 CW)
Probed H (160.8, 166.6):  1.571 (Fixed reference)
```

sandmmakers, 2024-03-15:
Was the bed temperature changing? Thermal expansion could account for at least .1mm of the change.

Toetje585, 2024-03-15:
> Was the bed temperature changing? Thermal expansion could account for at least .1mm of the change.

The bed and nozzle temp where stable, if I restart BedLever5000 without restarting the printer the Fixed Reference drops again.

Toetje585, 2024-04-01:
The issue was mechanical related... seems like one of the Shaft coupling was lose.... forgot to tight it I think 👍

closed 2024-04-01 by Toetje585