sandmmakers/BedLeveler5000 #61: Can't use BL5K with custom probe.

daniiooo, 2024-04-19:
The program fails at initialization because it's expecting either [bltouch] or [probe] in printer.cfg.
After connecting to the printer and homing the program throws an error.
I've replaced the stock elegoo probe with eddy current sensor [Cartographer3D](https://docs.cartographer3d.com/)
Printer details:
Model: Neptune 4 Plus
Firmware: 

- Clean Klipper v0.12.0-159-g2425a746 (project [OpenNept4une](https://github.com/OpenNeptune3D/OpenNept4une))
- Moonraker v0.8.0-327-g9447494b
- Fluidd v1.29.1

[image]
[Printer config file](https://github.com/sandmmakers/BedLeveler5000/files/15044627/printer.txt)

sandmmakers, 2024-04-19:
The fields in the `cartographer` section appear to match up with the required fields in the existing `probe` and `bltouch` sections. I should probably be able to add support without too much difficulty, although I'm not able to test.

daniiooo, 2024-04-19:
I can test the new functionality with cartographer. I remembered about this program recently after using it back when series 4 came out and want to try it out again with the new probe since the default induction one was pretty problematic.

sandmmakers, 2024-04-21:
I've started working on this, but can you please provide me with the output from the following:
1. Ensure Moonraker is running (e.g., verify Fluidd and/or Mainsail are working.)
2. Open a web browser
3. Enter the URL (replacing <PRINTER_IP> with the printer's IP address):
    <PRINTER_IP>/printer/objects/query?configfile
4. A large JSON response should be shown on the screen

daniiooo, 2024-04-21:
Here is the JSON output.
[query.json](https://github.com/sandmmakers/BedLeveler5000/files/15051749/query.json)
[query_formatted.json](https://github.com/sandmmakers/BedLeveler5000/files/15051750/query_formatted.json)

sandmmakers, 2024-04-21:
Awesome, thank you!

sandmmakers, 2024-04-21:
It appears that the Cartographer probe supports multiple "models". I was hoping that Klipper/Moonraker would merge the current/active model into the common Cartographer settings, but that doesn't appear to be the case. This means I will need to find a way to determine which model is currently active. While I'm still working on this task, I might require a bit more time than I initially expected.

daniiooo, 2024-04-23:
I've asked on the products discord about a way to maybe get that info through moonraker API but sadly the creator of the cartographer said that currently he doesn't think it's possible. He added it to the potential feature list though.

sandmmakers, 2024-04-25:
Thank you for the information. I'll keep working on this with the application just assuming the `default` mode is always active.

sandmmakers, 2024-05-01:
I'm still working on this. I'm planning on adding support for Cartographer and eddy probes (BTT Eddy) at the same time. My BTT Eddy probe arrived over the weekend and I already have it installed in a Neptune 3 Max (with Klipper) and working with BL5K.

daniiooo, 2024-06-14:
Sorry to bother but has there been any updates on the issue?

sandmmakers, 2024-06-15:
I'm still working on this ticket, but progress has been very slow. I installed an Eddy probe on my Neptune 3 Max (with Klipper) and spent a significant amount of time trying to get it to work, but given the number of bugs in Klipper's eddy probe support (and BTT's code changes), I've still been unable to get reasonable sensor readings. This item is currently blocked, waiting for eddy probe support in Klipper to mature.

daniiooo, 2024-06-15:
Thank you for the update. From what I know eddy from btt is the only one that's doing a native klipper approach for the handling of the probe and even then for now its on BTTs fork of klipper. I've heard cartographer is planning a native klipper support but thats way out and I don't know what beacon is planning. For now both carto and beacon have their own handling of the probing thats separate from klipper. That's just additional bits of information that i've been able to gather around. I'll be using the build in klipper leveling feature for now then and looking forward to the update of bed leveler in the future.

HutchMD, 2025-08-29:
Thanks for making the BedLeveler5000, 

I installed a BTT Eddy on my Neptune 4 Max, and I had the same error as daniiooo;
There's been more development in the community with eddy-ng.
I generally followed these steps in [modbot's video for eddy-ng with the BTT Eddy Duo/USB](https://www.youtube.com/watch?v=ImOpsdSRxt8).
I got the BTT Eddy working pretty well, but I was hoping to also use the BedLeveler5000 to adjust two knobs and ran into the probe error.

Unsure if this is helpful, but [here is the documentation on the Klipper config lines for the BTT Eddy for eddy-ng](https://github.com/vvuk/eddy-ng/wiki/BTT-Eddy)