sandmmakers/BedLeveler5000 #63: Incorrect number of tokens in response error when probing

QuikSilver156, 2024-05-11:
Attached log as requested thanks!
[log.txt](https://github.com/sandmmakers/BedLeveler5000/files/15284509/log.txt)

QuikSilver156, 2024-05-11:
Information on printer
Ender 3 (original not pro, max, etc...)
Installed bltouch
glass bed
running marlin bugfix 2.1.x  2024-02-28

sandmmakers, 2024-06-15:
Sorry, for not commenting earlier. Somehow I missed this ticket and only became aware of it now. I will look into the issue. Thank you for the log file!!

sandmmakers, 2024-06-15:
I've been able to reproduce the issue in the Marlin Simulator so it shouldn't be too difficult to fix.

QuikSilver156, 2024-06-16:
That's great news! Can't wait to try it out.

sandmmakers, 2024-06-16:
I've actually already fixed the problem. The problem is/was occurring due to a bug in Marlin (the failing response line ending with `\r\n` instead of the correct `\n`) but I was able to add a workaround so BL5K now supports both line ending formats. Since I'm currently cleaning up a number of little issues in preparation for a new bug fix release, I haven't pushed the change to Github yet, but I will soon.

closed 2024-06-18 by sandmmakers