sandmmakers/BedLeveler5000 #65: Connect to Creality CR-10 V3 causes firmware reboot

BMWStew, 2024-10-29:
I can connect to my CR-10 V3 using USB serial cable and do so with Visual Studio Code to upload new compiled firmware. Currently I use TH3D Studio Unified Firmware 2, version 2.87. I have installed BL Touch and filament runout sensor, too. Any time I attempt connection via USB serial to my printer and BL5K or the printer wizard, the printer reboots. I've tried different baud rates to no avail.  I can connect via PuTTY or Pronterface using 115200/none/8/1.

sandmmakers, 2024-12-22:
Can you please attach a log file? Thank you.

BMWStew, 2024-12-22:
Not any longer since I ended up selling that printer. But I appreciate your response. Please close the ticket.

Stew

sandmmakers, 2024-12-22:
Sorry for the delay, I had to take a break from 3d printing for a while. If you encounter any other issues with BL5K, please let me know.

closed 2024-12-22 by sandmmakers