sandmmakers/BedLeveler5000 #66: Not necessarily an issue

jStrahan73, 2024-12-16:
When ran it connects to printer then gives error saying no sensor found and exits

sandmmakers, 2024-12-22:
Can you please provide more details? What printer model? Is the printer stock? Can you please provide a log file?

jStrahan73, 2024-12-22:
Sorry thought I posted model. It’s a qidi q1 pro. Yes it is stock. I can get a log for you later

sandmmakers, 2024-12-27:
I just checked and that is a fairly recent Klipper-based printer. I suspect it might a a currently unsupported probe. Please include a copy of your `printer.cfg` with the log. Eddy probe, beacon, and similar are currently not supported, but I would like to add support in the future. When I originally looked into it, (mainline) Klipper support was still in flux, so it didn't make sense to spend too much time on it.