sandmmakers/BedLeveler5000 #68: No port selection possible

Chief-Karsten, 2025-01-05:
After I saw this great tool on YouTube and finally found it here on GitHub, I wanted to try it out today.

So I put the laptop in the basement, connected the USB cable to the Elegoo Neptune 3 Plus and started the Bed Leveler 5000.
In the program I selected the above-mentioned printer and tried to select a port to connect as shown in the video.

However, no selection of different ports appears - so I can't connect. I've already tried all the available USB ports - no success.

What did I do wrong?

sandmmakers, 2025-01-17:
Are you using Windows 10 or 11? If so, you can verify if the OS is detecting the printer correctly by following the steps below:
1. Type `Device Manager` into the **Windows Search** bar
2. Click on the `Device Manager` entry that appears
3. Expand the **Ports (COM & LPT)** section of the device list that appears
4. You should see an entry for your printer

If you don't see an entry, then there is an issue unrelated to BL5K.

Chief-Karsten, 2025-01-19:
Thanks for the answer.

I'm currently using Windows 10.
I'll try it out later today and let you know if it works - or not.

Thanks again for the help and have a nice day.

Greetings from Germany.

Chief-Karsten, 2025-02-12:
Sorry for the late reply. Things have been hectic privat. 
In the meantime, I was able to connect my printer and also perform the bed leveling.

Thank you for asking.